JdeRobot/BehaviorMetrics #19: DRL Cartpole example

igarag, 2020-03-16:
For the initiation of reinforcement learning with the Gym-Gazebo library and OpenAI Gym

SakshayMahna, 2020-03-22:
Hello,
I would like to work on it!
I am thinking of adding a new directory in the main repo, would this be right? The directory will contain the OpenGym environment configuration and an RL algorithm that is applied on the environment.

fqez, 2020-03-23:
Hi @SakshayMahna, I'm currently working on a reboot for this project, so all the contents are most likely prone to disappear soon (I hope by the end of this week, or the beginning of the next). So I wouldn't recommend working on this repository yet.

I will let you know when the first "stable" version of the reboot is ready, so you all can start contributing.
Regards!

dcharrezt, 2020-06-14:
Hi @igarag,

I would like to help with this issue, is it going to be added to a new tab similar to the [Jderobot Academy exercises](http://jderobot.github.io/RoboticsAcademy/exercises/)?

igarag, 2020-06-14:
Hello.

The last thing we managed to do with this incident was to launch the world of Gazebo with the Cartpole models but without success to be able to move the car or the stick.

The models can be found in the 'erlerobot' repository at [this link](https://github.com/erlerobot/cartpole_gazebo). You can try to make it work externally until we have clear the location of the learning-by-doing exercises.

If the exercise is successfully ported it can become part of the JdeRobot Academy exercise collection, of course, as long as we accommodate the platform infrastructure in the near future: Noetic with Python3, etc...