JdeRobot/BehaviorMetrics #20: Docs website

igarag, 2020-03-27:
Firtst version of web documentation based on minimal mistakes template.

yashagarwal9, 2020-03-31:
Hello, @igarag I am interested in working on this and want to start with the installation first. Can you please help me with this?

SakshayMahna, 2020-03-31:
Hello!
The website uses [Jekyll](https://jekyllrb.com/) along with [Minimal Mistakes theme](https://mmistakes.github.io/minimal-mistakes/) to render the webpages. The instructions given [here](https://github.com/JdeRobot/BehaviorSuite/tree/master/docs) in the docs folder of repository can be followed to install and run the jekyll server.

igarag, 2020-03-31:
Hi, as @SakshayMahna  says the template for creating pages with github.io you can find it in the [`docs` section of the repository](https://github.com/JdeRobot/BehaviorSuite/commit/a96e1b5518ed40d36e0895afd79d6366286166f6). You can find instructions for installing Jekyll on your machine in [this JdeRobot repository](https://github.com/JdeRobot/jderobot.github.io/blob/master/README.md).

igarag, 2020-03-31:
The aim of this issue is to move the documentation found in the project's README's to the documentation page. In this week (or next week at the most) we are going to introduce a reboot in the project that will change the structure and approach.

Please wait a few weeks until this process is completed. Once it's over we'll put in place a **manual for the contribute**.

Thank you.

yashagarwal9, 2020-03-31:
@igarag I have installed the Jekyll and ran the server on my machine. By the time reboot is introduced I just try adding the content for the documentation.

Thank you.

igarag, 2020-06-11:
@yashagarwal9 Sorry I'm delayed. Now that the paperwork is up to date, are you still having the same problem?

closed 2020-10-21 by sergiopaniego