JdeRobot/BehaviorMetrics #21: error while running classification network, instructions aren't clear

itsyashkothari, 2020-03-27:
1. Are the models provided in ReadMe trained or we have to first train then and run? Are they classification network or regression?
2. What is manual usage? Does that mean a way to try our solutions?
3. Running driver.py gives module not found error for PyQt5, but it isn't available for python 2.7, what to do ?

Currently I have unzipped the provided models in net/models/ and dataset in net/dataset/ and just left 2017-tfm-vanessa-fernandez in the repo.

Korusuke, 2020-03-27:
> 1. Are the models provided in ReadMe trained or we have to first train then and run? 

The models provided are trained.

> Are they classification network or regression?

You can identify that by their names and cross-referring to `classification_model.py` and `regression_model.py`.

For example a model with file name as : `model_smaller_vgg_7classes_balanced_cropped_w.h5` is a classification network using smaller vgg net implementation.

**Note**: Don't forget to set `Cropped: True` in driver.yml

> 3. Running driver.py gives module not found error for PyQt5, but it isn't available for python 2.7, what to do ?

I am not sure about this, as I don't have PyQt5 installed and it's working for me, can you give an error log?

itsyashkothari, 2020-03-27:
Thanks for the reply.

```shell
(jde) yash@jarvis:~/projects/gsoc/jde/BehaviorSuite/vision-based-end2end-learning$ python driver.py driver.yml     
Traceback (most recent call last):  
  File "driver.py", line 34, in <module>  
    from gui.GUI import MainWindow  
  File "/home/yash/projects/gsoc/jde/BehaviorSuite/vision-based-end2end-learning/gui/GUI.py", line 7, in <module>  
    from PyQt5 import QtGui  
ImportError: No module named PyQt5  
```

Korusuke, 2020-03-27:
I am not sure but try running `sudo apt-get install python-pyqt5` and then try again.

itsyashkothari, 2020-03-27:
Its showing already installed.

More info - I'm running this using python 2.7 as mentioned, can you confirm your version? 
I built PyQt5 from source but then i got following error 
``` shell 
ImportError: /usr/lib/x86_64-linux-gnu/libQt5Core.so.5: version `Qt_5.12' not found    
(required by /home/yash/anaconda3/envs/jde/lib/python2.7/site-packages/PyQt5/QtGui.so)
```
apt version installed 5.10.1 and source code is 5.12, so maybe issue because of that.
Maybe i'm missing something very obvious because others are able to run the code.

Korusuke, 2020-03-27:
> Its showing already installed.
> 
> More info - I'm running this using python 2.7 as mentioned, can you confirm your version?

`Python 2.7.17`

> I built PyQt5 from source but then i got following error
>
> ```shell
> ImportError: /usr/lib/x86_64-linux-gnu/libQt5Core.so.5: version `Qt_5.12' not found
> (required by [...]

Compiling from source should not be required since apt is already installing it.

Can you try using normal python instead of anaconda?
I am not sure but that might be the issue.

Korusuke, 2020-03-27:
I searched a bit more and I guess, you can either switch to linux- python and it will work since linux python can access apt packages but conda cannot

or

install pyqt in conda using `conda install -c anaconda pyqt` [page](https://anaconda.org/anaconda/pyqt)

Hope this works.

itsyashkothari, 2020-03-27:
> I searched a bit more and I guess, you can either switch to linux- python and it will work since linux python can access apt packages but conda cannot

This maybe it. I was doing inside conda env. 
Is pyqt same as pyqt5? looks different.
python2 doesn't have pyqt5 so can't be installed, I switched to python3 and installed pyqt5, now its working great. 
Thanks for the help.

closed 2020-03-27 by itsyashkothari

jmplaza, 2020-03-27:
Folks, @itsyashkothari @Korusuke BehaviorSuite is a work-in-progress project. It is still under construction.