JdeRobot/BehaviorMetrics #24: No module named PyQt5.QtWidgets (python2.7)

dcharrezt, 2020-06-08:
Hi, 

I was trying to run the `driver.py` GUI but I kept getting this error.

```bash
[Behavior-Log] - INFO: GazeboEnv: launching gzserver.
[Behavior-Log] - INFO: Pausing simulation
[Behavior-Log] - INFO: Executing app
[Behavior-Log] - ERROR (driver: 138): No module named PyQt5.QtWidgets
[Behavior-Log] - INFO: closing all processes...
[Behavior-Log] - INFO: DONE! Bye, bye :)
```

I tried to install PyQt5 as mentioned in the readme using apt install

```bash
sudo apt-get install python-pyqt5  
sudo apt-get install pyqt5-dev-tools
sudo apt-get install qttools5-dev-tools
```
and this [gist](https://gist.github.com/rllola/da97b890a79d206f00a909bc2a56598d) but did not work.

fqez, 2020-06-11:
Hi @dcharrezt 

There are updated instructions here (https://jderobot.github.io/BehaviorStudio/install/) about pyqt5 dependencies for python2. Check it out and let me know if those are enough to make the application work, please.
In case that they are not, here are all the pyqt5 packages installed in a clean virtual machine. These make the application work as intended:

```
libqscintilla2-qt5-13/bionic,now 2.10.2+dfsg-4 amd64 [installed,automatic]
libqscintilla2-qt5-l10n/bionic,bionic,now 2.10.2+dfsg-4 all [installed,automatic]
libqt5concurrent5/bionic-updates,bionic-security,now 5.9.5+dfsg-0ubuntu2.5 amd64 [installed,automatic]
libqt5core5a/bionic-updates,bionic-security,now 5.9.5+dfsg-0ubuntu2.5 amd64 [installed,automatic]
libqt5dbus5/bionic-updates,bionic-security,now 5.9.5+dfsg-0ubuntu2.5 amd64 [installed,automatic]
libqt5designer5/bionic,now 5.9.5-0ubuntu1 amd64 [installed,automatic]
libqt5designercomponents5/bionic,now 5.9.5-0ubuntu1 amd64 [installed,automatic]
libqt5gui5/bionic-updates,bionic-security,now 5.9.5+dfsg-0ubuntu2.5 amd64 [installed,automatic]
libqt5help5/bionic,now 5.9.5-0ubuntu1 amd64 [installed,automatic]
libqt5network5/bionic-updates,bionic-security,now 5.9.5+dfsg-0ubuntu2.5 amd64 [installed,automatic]
libqt5opengl5/bionic-updates,bionic-security,now 5.9.5+dfsg-0ubuntu2.5 amd64 [installed,automatic]
libqt5opengl5-dev/bionic-updates,bionic-security,now 5.9.5+dfsg-0ubuntu2.5 amd64 [installed,automatic]
libqt5positioning5/bionic,now 5.9.5+dfsg-0ubuntu2 amd64 [installed,automatic]
libqt5printsupport5/bionic-updates,bionic-security,now 5.9.5+dfsg-0ubuntu2.5 amd64 [installed,automatic]
libqt5qml5/bionic-updates,now 5.9.5-0ubuntu1.1 amd64 [installed,automatic]
libqt5quick5/bionic-updates,now 5.9.5-0ubuntu1.1 amd64 [installed,automatic]
libqt5quickparticles5/bionic-updates,now 5.9.5-0ubuntu1.1 amd64 [installed,automatic]
libqt5quicktest5/bionic-updates,now 5.9.5-0ubuntu1.1 amd64 [installed,automatic]
libqt5quickwidgets5/bionic-updates,now 5.9.5-0ubuntu1.1 amd64 [installed,automatic]
libqt5sensors5/bionic,now 5.9.5-0ubuntu1 amd64 [installed,automatic]
libqt5sql5/bionic-updates,bionic-security,now 5.9.5+dfsg-0ubuntu2.5 amd64 [installed,automatic]
libqt5sql5-sqlite/bionic-updates,bionic-security,now 5.9.5+dfsg-0ubuntu2.5 amd64 [installed,automatic]
libqt5svg5/bionic,now 5.9.5-0ubuntu1 amd64 [installed,automatic]
libqt5test5/bionic-updates,bionic-security,now 5.9.5+dfsg-0ubuntu2.5 amd64 [installed,automatic]
libqt5webchannel5/bionic,now 5.9.5-0ubuntu1 amd64 [installed,automatic]
libqt5webkit5/bionic,now 5.212.0~alpha2-7ubuntu1 amd64 [installed,automatic]
libqt5webkit5-dev/bionic,now 5.212.0~alpha2-7ubuntu1 amd64 [installed,automatic]
libqt5widgets5/bionic-updates,bionic-security,now 5.9.5+dfsg-0ubuntu2.5 amd64 [installed,automatic]
libqt5xml5/bionic-updates,bionic-security,now 5.9.5+dfsg-0ubuntu2.5 amd64 [installed,automatic]
libqwt-qt5-6/bionic,now 6.1.3-1 amd64 [installed,automatic]
libqwt-qt5-dev/bionic,now 6.1.3-1 amd64 [installed,automatic]
pyqt5-dev/bionic,bionic,now 5.10.1+dfsg-1ubuntu2 all [installed,automatic]
pyqt5-dev-tools/bionic,now 5.10.1+dfsg-1ubuntu2 amd64 [installed]
python-pyqt5/bionic,now 5.10.1+dfsg-1ubuntu2 amd64 [installed]
python-pyqt5.qsci/bionic,now 2.10.2+dfsg-4 amd64 [installed]
python-pyqt5.qtopengl/bionic,now 5.10.1+dfsg-1ubuntu2 amd64 [installed,automatic]
python-pyqt5.qtsvg/bionic,now 5.10.1+dfsg-1ubuntu2 amd64 [installed]
python-pyqt5.qtwebkit/bionic,now 5.10.1+dfsg-1ubuntu2 amd64 [installed,automatic]
python3-pyqt5/bionic,now 5.10.1+dfsg-1ubuntu2 amd64 [installed]
python3-pyqt5.qtsvg/bionic,now 5.10.1+dfsg-1ubuntu2 amd64 [installed]
qt5-assistant/bionic,now 5.9.5-0ubuntu1 amd64 [installed,automatic]
qt5-gtk-platformtheme/bionic-updates,bionic-security,now 5.9.5+dfsg-0ubuntu2.5 amd64 [installed,automatic]
qt5-qmake/bionic-updates,bionic-security,now 5.9.5+dfsg-0ubuntu2.5 amd64 [installed,automatic]
qt5-qmake-bin/bionic-updates,bionic-security,now 5.9.5+dfsg-0ubuntu2.5 amd64 [installed,automatic]
qt5-qmltooling-plugins/bionic-updates,now 5.9.5-0ubuntu1.1 amd64 [installed,automatic]
```

Most of them are installed while installing ROS and JdeRobot-base dependencies. It would be nice to port the application to python3, to make the dependencies installation easier through pip packages.

Let me know if you tackle the issue down :)

dcharrezt, 2020-06-12:
Hi @fqez,

I installed those libraries, but I keep geting the same error.

```bash
sudo apt-get install libqscintilla2-qt5-13 libqscintilla2-qt5-l10n libqt5concurrent5 libqt5core5a libqt5dbus5 libqt5designer5 libqt5designercomponents5 libqt5gui5 libqt5help5 libqt5opengl5 libqt5opengl5-dev libqt5positioning5 libqt5printsupport5 libqt5qml5 libqt5quick5 libqt5quickparticles5 libqt5quicktest5 libqt5quickwidgets5 libqt5sensors5 libqt5sql5 libqt5sql5-sqlite libqt5svg5 libqt5test5 libqt5webchannel5 libqt5webkit5 libqt5webkit5-dev libqt5widgets5 libqt5xml5 libqwt-qt5-6 libqwt-qt5-dev pyqt5-dev pyqt5-dev-tools python-pyqt5 python-pyqt5.qsci python-pyqt5.qtopengl python-pyqt5.qtsvg python-pyqt5.qtwebkit python3-pyqt5 python3-pyqt5.qtsvg qt5-assistant qt5-gtk-platformtheme qt5-qmake qt5-qmake-bin qt5-qmltooling-plugins
```

That appears when I try to import PyQt5..QtWidgets

```bash
Python 2.7.17 (default, Apr 15 2020, 17:20:14) 
[GCC 7.5.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import PyQt5.QtWidgets
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named PyQt5.QtWidgets
```

I definitely would like to port it to python3!

sergiopaniego, 2020-06-21:
I'm having the same problem reported by @dcharrezt when trying running the GUI in python2
Is there any known solution?

jmplaza, 2020-06-21:
We are moving to Python3 (and ROS Noetic), aren't we?. Maybe we should focus on having the Formula1 race car at JdeRobot-base running fine, and then the whole BehaviorStudio could be in Python3...

dcharrezt, 2020-06-21:
Yes @jmplaza I should have closed this issue!.

closed 2020-06-21 by dcharrezt