JdeRobot/BehaviorMetrics #43: [Cleaning] The brains module has legacy unused functions

fqez, 2020-06-11:
There are several functions regarding the sensors of the robot that are not used and come from previous version of the application. Those should be removed.

parthskansara, 2020-10-04:
I'll work on it. Any tips on how to go about it?

sergiopaniego, 2020-10-06:
Have you managed to install the application in your computer?
Follow this guide (preferably Noetic part since it's the most updated part): https://jderobot.github.io/BehaviorStudio/install

After that the functions can be reviewed finding the ones that are no longer needed.

If you have the app installed, we can talk further!

pawanw17, 2021-01-28:
Hello @sergiopaniego I have successfully installed BehaviourStudio and would like to work on this issue.

sergiopaniego, 2021-01-29:
Second step after installing is to follow the quick start guide to launch your first brain https://jderobot.github.io/BehaviorStudio/quick_start/. If you find any error you can report it via issue.

Then you can start exploring the brain module to understand which code is not used and can be removed.

prateekkaistha, 2021-02-12:
After successfully installing BehaviourStudio and executing the first command from
[https://jderobot.github.io/BehaviorStudio/quick_start/](url) 
i.e, `python driver.py -c default.yml -g` 
it throws an error
[image]

prateekkaistha, 2021-02-14:
> After successfully installing BehaviourStudio and executing the first command from
> [https://jderobot.github.io/BehaviorStudio/quick_start/](url)
> i.e, `python driver.py -c default.yml -g`
> it throws an error
> [image]
> using ROS noetic in **Ubuntu 20.04**

The issue is resolved now and gazebo is loading as well after executing
`python3 driver.py -c default.yml -g`
found nearly at the bottom of the installation page [https://jderobot.github.io/BehaviorStudio/install/ros_noetic](url) XD

sergiopaniego, 2021-02-17:
@prateekkaistha What was the issue there? If something is missing in the installation recipe you could open a PR with the changes needed

prateekkaistha, 2021-02-21:
Sorry for the late reply,
Actual problem was with the installation of Behavior studio itself where make commands required sudo privileges and error thrown got unnoticed as commands were sequenced to run one after another
Just putting sudo before make commands would solve the issue

das-ankur, 2022-12-09:
If anyone isn't working on this issue. I would like to work on this issue.

sergiopaniego, 2022-12-12:
Hi @das-ankur! Thanks for your interest in contributing to this project and specifically to this issue. 
First of all, have you managed to install the application? 
This is an old issue but it's always interesting to do some refactor!

ChinmayMundane, 2023-02-24:
hi! I would like to work on this issue.
I tried installing the application using link given in above conversation but the link is broken.
Could you provide me with new link to install ?

nikitashrivastava29, 2023-03-07:
> Have you managed to install the application in your computer? Follow this guide (preferably Noetic part since it's the most updated part): https://jderobot.github.io/BehaviorStudio/install
>
> After [...]

Hey @sergiopaniego the link referencing to the installation of Behavior Studio is leading to a Page Not Found Error. Could you link the new page for the installation guide? It'd be really helpful!

tejalbarnwal, 2023-03-07:
Hey @sergiopaniego , I managed to install behavior metrics.
As stated in the [requirements.txt](https://github.com/JdeRobot/BehaviorMetrics/blob/noetic-devel/requirements.txt), `carla-birdeye-view 1.1.1` depends on `numpy==1.18.3`, but other modules like `tensorflow-gpu 2.11.0` depend on `numpy>=1.20`. Is there any way to upgrade the numpy dependency for carla-birdeye-view.

sergiopaniego, 2023-03-13:
Hi @nikitashrivastava29, the new link to the installation instructions is `https://jderobot.github.io/BehaviorMetrics/install/`. If you find any issue, please open an issue so we can discuss it.

sergiopaniego, 2023-03-13:
> Hey @sergiopaniego , I managed to install behavior metrics. As stated in the [requirements.txt](https://github.com/JdeRobot/BehaviorMetrics/blob/noetic-devel/requirements.txt), `carla-birdeye-view [...]

Hi @tejalbarnwal! Since carla-birdeye-view seems to be no longer in development, some packages could be old versions. I have tested with newer versions of numpy and they don't crash at any point, so it seems legit installing the latest numpy version. To overcome this issue during the installation, you can add --no-deps to the pip install -r requirements.txt command. Could you open a new issue and pull request with the updated command in the installation instructions? This could be a good starting point to start contributing! :smile:

tejalbarnwal, 2023-03-13:
Sure, I will submit a PR in some time. I have opened a new issue catering to this [here(https://github.com/JdeRobot/BehaviorMetrics/issues/584)

closed 2024-02-15 by sergiopaniego