JdeRobot/BehaviorMetrics #48: [Doc] *Add how to contribute* and *programming a brain* sections

fqez, 2020-06-11:
This two sections should appear in the documentation since is the gateway to new developers.

yashasvimisra2798, 2020-12-19:
I would like to work on this issue. May you please elaborate on what exactly am I supposed to add under the *programming a brain* section.

sergiopaniego, 2020-12-23:
Hi! Sorry for the late response. 
The `programming a brain` section should contain instructions on how to generate your own python brain that can control the robot and try to complete the circuit. 

First of all, I would suggest installing the project and try running it. Have you already done it? For further instructions you can follow [this link](https://jderobot.github.io/BehaviorStudio/install/ros_noetic). Only ROS Noetic is supported at the moment.

laukikk, 2022-04-18:
The documentation for `how to contribute` is already present [here](https://jderobot.github.io/BehaviorMetrics/documentation/contributing/). 
So, now only the `programming a brain` section is left. I will start working on that now. As the [original link](https://jderobot.github.io/BehaviorMetrics/quick_start/tutorial/) to access the docs is present in the [QuickStart](https://jderobot.github.io/BehaviorMetrics/quick_start/) docs, I would add add the docs there itself.

laukikk, 2022-05-08:
@sergiopaniego, I feel even this issue can be closed now.

laukikk, 2022-05-18:
> @sergiopaniego, I feel even this issue can be closed now.

@sergiopaniego