JdeRobot/BehaviorMetrics #82: Error in launching Gazebo

adityarajsahu, 2020-09-29:
[image]

When I run the command "python driver.py -c default.yml -g" in the terminal, I am getting this error. How to remove it?

sergiopaniego, 2020-09-29:
Could you please give more details about this issue?
Are you running Noetic or Melodic?

adityarajsahu, 2020-09-29:
I am running melodic

sergiopaniego, 2020-09-29:
We're currently just offering support for ROS Noetic because the project is pretty new and we have to prioritise resources 😄 . The latest updates are in [noetic-devel branch](https://github.com/JdeRobot/BehaviorStudio/tree/noetic-devel) if you want to have a look. 

You can find the installation guide [here](https://jderobot.github.io/BehaviorStudio/install/ros_noetic). We have two different installations, native and using docker. If you want to get started quickly with the tool I suggest you to go directly to the docker installation section [here](https://jderobot.github.io/BehaviorStudio/install/ros_noetic#docker-installation). This way you can download the image and run it and you will have the environment all setup.

Please let us know if you manage to complete the installation!

fqez, 2020-10-04:
Hi @adityarajsahu,

it seems like the application can't locate the resources file. Did you run pyrcc? In the root directory run the following:

`pyrcc5 -o ui/gui/resources/resources.py ui/gui/resources/resources.qrc`

and try to launch the app again as you did. Hope it helps!

closed 2020-10-26 by adityarajsahu