JdeRobot/BehaviorMetrics #85: Information about a repository

adityarajsahu, 2020-10-02:
Please anyone explain me about this repository: https://github.com/RoboticsLabURJC/2017-tfm-vanessa-fernandez.git 

It has been mentioned to clone this repository in the README.md file of **BehaviorStudio/behavior_studio**.

sergiopaniego, 2020-10-02:
This project (BehaviorStudio) began from that repo (a final master thesis), so some dependecies could still be attached to that one. We're working on detach  both repositories, but as I pointed in a previous issue, we are now only supporting ROS noetic (noetic-devel branch).

You could try avoiding downloading that repo and try running the project. If you encounter any issue, we can further give you details.

sergiopaniego, 2020-10-02:
You could find useful Noetic installation guide (https://jderobot.github.io/BehaviorStudio/install/ros_noetic) since the project should look exactly the same and be run on a similar base. 

The README that you're using is not updated with latest information

adityarajsahu, 2020-10-02:
So, I don't need to do any installations mentioned in the BehaviorStudio/behavior_studio/README.md file ?

sergiopaniego, 2020-10-02:
You'll probably need JdeRobot-base, JdeRobot-assets (https://github.com/JdeRobot/BehaviorStudio/tree/melodic-devel/behavior_studio#installation) and ros-melodic-gazebo-plugins from that part. 

JdeRobot-assets is currently being deprecated in favor of [CustomRobot](https://github.com/JdeRobot/CustomRobots) so it won't be needed in the near future.

I still recommend you using [ROS Noetic](https://jderobot.github.io/BehaviorStudio/install/ros_noetic) natively or using the docker image we created to quickly get started, since that's the part that is currently more advanced and more stable. It will be easier for you to get started easily. The installation for Melodic is not updated and could lead to some problems.

Hope this helps!

fqez, 2020-10-04:
Hi! 

The README is outdated. Actually there's an issue open about that matter in #49. Try following the steps in here https://jderobot.github.io/BehaviorStudio/install/ros_melodic and let us know if you face any issues.

Sorry about the documentation, I'm kinda busy lately, but I will come back by the end of this month to try and stabilize this repo.

closed 2020-10-26 by adityarajsahu