JdeRobot/BehaviorMetrics #105: Error in the command "python2 driver.py driver.yml"

adityarajsahu, 2020-10-24:
[image]

While running the command "python2 driver.py driver.yml", I got this error. Moreover, I can't find "driver.yml" anywhere in the repository.

jmplaza, 2020-10-24:
python2 ??? AFAIK we work only with python3 in BehaviorStudio...

sergiopaniego, 2020-10-24:
Behavior Studio uses python3. 
You can find driver.yml as default.yml in the repo. It's the default configuration to run the program. You can also find information about it on the web under Quick start section.

jmplaza, 2020-10-25:
Hi @adityarajsahu I suggest you better work on noetic-devel branch, even using a docker file with Noetic on your local host regardless your Linux release. Noetic is the default infrastructure for BehaviorStudio. BehaviorStudio is NOT SUPPORTED in Melodic. In Melodic python2 is the default Python release. We are few developers and don't have enough time to give support to Melodic branch, we are just focused on the tool at Noetic and using Python3.

jmplaza, 2020-10-25:
Fixed in #106

closed 2020-10-25 by jmplaza

adityarajsahu, 2020-10-26:
@jmplaza I currently have ubuntu 18.04 LTS installed in my laptop. Will I face any problem if I clone noetic-devel branch in it?

jmplaza, 2020-10-26:
The default ROS distribution for Ubuntu 18.04 LTS is Melodic. BehaviorStudio does not work with Melodic, it works with Noetic instead. My suggestion is you install a Ubuntu 20.04 LTS inside a docker container, install that docker image on your 18.04 and work inside that docker. There the ROS distribution will be Noetic, the default Python release will be Python3 and BehaviorStudio will work there, up to date with its master branch. 

We created a docker image for this. Check the instructions [here](https://jderobot.github.io/BehaviorStudio/install/ros_noetic#docker-installation) https://jderobot.github.io/BehaviorStudio/install/ros_noetic#docker-installation