JdeRobot/BehaviorMetrics #140: Introduce Continuous Control F1RL Environment

UtkarshMishra04, 2021-03-16:
Current Setup:
DQN and Q-learning are already done with Discrete Control Actions

To be introduced:
Policy Gradient Algorithm with Continuous Control Actions

Major Changes:
1. Add a continuous control environment derived from DQN environment by Change the action space
2. Add a brain to the f1rl module to execute the actions

UtkarshMishra04, 2021-03-16:
I have created the files mentioned above and have started testing the environment file.

However, I get the rosmaster error:
`[ERROR] [1615899275.259326]: Unable to immediately register with master node [http://localhost:11311]: master may not be running yet. Will keep trying.`
which is not the case for any different brain.

This is my configuration file. PS: I have disabled the algorithm parameters (aka robot['Parameters']) in `utils/configuration.py` temporarily.

> Behaviors:
    Robot:
        Sensors:
            Cameras:
                Camera_0:
                    Name: 'camera_0'
                    Topic: '/F1ROS/cameraL/image_raw'
            Pose3D:
                Pose3D_0:
                    Name: 'pose3d_0'
                    Topic: '/F1ROS/odom'
        Actuators:
            Motors:
                Motors_0:
                    Name: 'motors_0'
                    Topic: '/F1ROS/cmd_vel'
                    MaxV: 3
                    MaxW: 0.3
        BrainPath: 'brains/f1rl/f1_follow_line_camera_continuous.py'
        Type: 'f1rl'
    Simulation:
        World: /opt/jderobot/share/jderobot/gazebo/launch/f1_1_simplecircuit.launch
    Dataset:
        In: '/tmp/my_bag.bag'
        Out: ''
    Stats:
        Out: './'
        PerfectLap: 'lap-simple-circuit.bag'
    Layout:
        Frame_0:
            Name: frame_0
            Geometry: [1, 1, 2, 2]
            Data: rgbimage
        Frame_1:
            Name: frame_1
            Geometry: [0, 1, 1, 1]
            Data: rgbimage

sergiopaniego, 2021-03-17:
If you're able to run any other brain under `f1rl` folder, I guess that the problem must come from your brain file `f1_follow_line_camera_continuous.py`

The simulation world starts?
Have you followed the same structure as the other brains in the folder? 
I will need more details to help you 😄

UtkarshMishra04, 2021-03-17:
Hi
I forgot to mention that all f1rl brain is giving the same error.
The simulation world is not starting but Behavior studio window opens up.
Yes, my structure is perfectly the same as that of other brains, in particular with the DQN brain.