JdeRobot/BehaviorMetrics #158: Hard-Coded and Unnecessary segments should be removed

UtkarshMishra04, 2021-04-08:
Current Setup:
There is a lot of hard coding which brings a lot of problems when trying to generalize the framework across multiple settings.

To be introduced:
Code will be cleaned and hardcoded segments will be removed, proper parameterization in Configs will be used for all the variables and hyperparameters.

Major Changes:
The overall task of running custom settings with the default config template will be much easy after this.

This will also resolve #43

sergiopaniego, 2021-05-07:
I believe we should deeply review the RL brains removing any unnecessary code and generalising them to easily train-test the agents. What do you think? It's important to already consider this issue for #157! We could add a table in the documentation (Github pages webpage) with the information about available brains and how to train-test them. I thinks it's kind of difficult to understand the RL brains now. Thoughts?

sergiopaniego, 2021-05-13:
With this changes that you suggest, the DQN implementation would be just in PyTorch?
We need to still support Tensorflow-Keras for the non-RL part since all the brains use it.