JdeRobot/BehaviorMetrics #159: ImportError: Cannot import name 'transpose_shape' from 'keras.utils.generic_utils'

predator4hack, 2021-04-10:
Expected behavior: Keras should be imported normally
Actual behavior: On importing keras, it gives an import error: `Cannot import name 'transpose_shape' from 'keras.utils.genric_utils'`
Steps to reproduce:
$python3
>>>import keras

sergiopaniego, 2021-04-11:
* Which file is giving that error?
* Could you share more details about how to reproduce?

predator4hack, 2021-04-11:
I was trying to run f1_follow_line_camera_dqn.py brain, so I changed the brain path in default-rl.yml. But when I executed `python3 driver.py -c default-rl.yml -g` that error shows up on line `from keras import backend as K`. On further investigation, I tried to import keras on python3 shell, and that was showing up the same error.

Also, there were some warnings: `Using modular system for hdf5, please switch to tensorflow-io`

sergiopaniego, 2021-04-12:
Could you share your full `default-rl.yml`?

predator4hack, 2021-04-29:
Sorry for the late reply, I was occupied with my end-semester examinations
These are the contents of default-rl.yml file :
`Behaviors:
    Robot:
        Sensors:
            Cameras:
                Camera_0:
                    Name: 'camera_0'
                    Topic: '/F1ROS/cameraL/image_raw'
            Pose3D:
                Pose3D_0:
                    Name: 'pose3d_0'
                    Topic: '/F1ROS/odom'
        Actuators:
            Motors:
                Motors_0:
                    Name: 'motors_0'
                    Topic: '/F1ROS/cmd_vel'
                    MaxV: 3
                    MaxW: 0.3
                    RL: True
        BrainPath: 'brains/f1rl/f1_follow_line_camera_dqn.py'
        Type: 'f1rl'
        Parameters:
            action_set: 'hard'
            gazebo_positions_set: 'pista_simple'
            alpha: 0.2 
            gamma: 0.9
            epsilon: 0.99
            total_episodes: 20000
            epsilon_discount: 0.9986 
            env: 'camera'
    Simulation:
        World: /opt/jderobot/share/jderobot/gazebo/launch/simple_circuit.launch
    Dataset:
        In: '/tmp/my_bag.bag'
        Out: ''
    Stats:
        Out: './'
        PerfectLap: 'lap-simple-circuit.bag'
    Layout:
        Frame_0:
            Name: frame_0
            Geometry: [1, 1, 2, 2]
            Data: rgbimage
        Frame_1:
            Name: frame_1
            Geometry: [0, 1, 1, 1]
            Data: rgbimage
        Frame_2:
            Name: frame_2
            Geometry: [0, 2, 1, 1]
            Data: rgbimage
        Frame_3:
            Name: frame_3
            Geometry: [0, 3, 3, 1]
            Data: rgbimage`

GloryyrolG, 2021-05-04:
Using pip install solves my similar problem where I installed Keras, etc., by Conda dependencies.

sergiopaniego, 2021-05-07:
Can't reproduce the error. It works fine for me using that config file. Could you download latest docker version with dependencies and test that it works?

Issue closed.

closed 2021-05-07 by sergiopaniego

supriyaarcot, 2022-04-01:
Expected behavior: Keras should be imported normally
Actual behavior: On importing keras, it gives an import error: Cannot import name 'transpose_shape' from 'keras.utils.genric_utils'

sergiopaniego, 2022-04-01:
Are you using the versions stated in https://github.com/JdeRobot/BehaviorMetrics/blob/noetic-devel/requirements.txt ?