JdeRobot/BehaviorMetrics #250: [F1] Spline Interpolation to calculate Position Deviation Metric

SakshayMahna, 2021-11-18:
We can experiment with the following approach for calculating Position Deviation for Formula 1 Car metrics:

- Calculate a Cubic Spline between the checkpoints of perfect lap
- From the current point in consideration, calculate the minimum distance point on spline
- Add the difference to the Position Deviation list

This can help get more accurate results

sergiopaniego, 2021-11-18:
Good idea!
How about computation time?
I think we could implement a small draft to check if the approach could improve the results. 
Could you generate this draft @SakshayMahna ?

SakshayMahna, 2021-11-19:
Computation Time may depend on the implementation, I guess.
I'll start with a draft and let's see the results.

SakshayMahna, 2021-11-22:
I have created a draft implementation. It uses two different splines for `x` and `y`, both parameterized by a separate variable `t`.

Computation wise, the method doesn't take too much time to generate results.

I tried running it with the `explicit_brain`, here are the results.

[image]
*Stats using Spline Implementation*

[image]
*Stats using previous to previous implementation*

SakshayMahna, 2021-11-23:
These are the results for PilotNet and LSTM:

[image]
*PilotNet*

[image]
*LSTM*

There seems to be some problem, both the PilotNet and LSTM deviations are less than that of Explicit one, and LSTM error is lower than that of PilotNet.

sergiopaniego, 2021-11-23:
Those results are kind of weird.
Do you have any idea of where the problem could be located? 
Can we easily weight the different errors? This option could make the latter results show the real difference. We would add some factor of multiplication to the minimum error for each point so if the error is high for some time step, that error adds more importance to the final calculation. This is something that I'm currently experimenting with for the previous implementation.

SakshayMahna, 2021-11-23:
Two points I can think of, to check for errors:

- Have a look at what is the shape of spline and how the points of the generated ROS bag are placed with respect to it, using some plots
- General look at the print statements of the calculations

Yes, we can easily weight the different errors. This snippet is still the same in the implementation:

```python
if min_dist < 100:
    min_dists.append(min_dist)
```

sergiopaniego, 2021-11-23:
Yep, the code is still the same on noetic-devel branch. It's on this PR: https://github.com/JdeRobot/BehaviorMetrics/pull/253

SakshayMahna, 2021-11-25:
One error I could identify: I was not calculating the square root of the distance.

After fixing that, and making the changes according to PR: #253 . These are the results:

[image]
*Explicit Brain*

[image]
*Pilotnet Brain*

The values come out to be quite large, due to the exponentiation. However, they seem to reflect the difference. Does it look correct to you @sergiopaniego?

sergiopaniego, 2021-11-25:
My results with the current implementation are the following for simple_circuit:
*Explicit brain*
[image]
*Pilotnet brain*
[image]

Are you applying the multiplication factor for each error? Is the PR updated with the latest improvements?

Your results seem fine but there is some difference to the ones that I obtained.

SakshayMahna, 2021-11-26:
After some debugging and experiments. I'll summarize all the results/observations with the plots:

1. There are 2 ways to calculate the difference, taking the Path(Generated Checkpoints) as Spline or taking the Track(Perfect Checkpoints) as Spline. I found that the perfect checkpoints are not distributed in a smooth way and it is difficult to fit a Cubic Spline through it. I think that using Path as Spline would be good due to the smoothness property and the next point well.

[image]
*Track (Perfect Checkpoints) as Spline*

[image]
*Path (Generated Checkpoints) as Spline*

2. The Perfect Checkpoints do not seem to be distributed in a manner similar to how our car moves on the circuit. I plotted the first 100 points, last 100 points and the last point(see the yellow points). Due to this it is difficult to fit a spline directly to these points.

[image]
*Last point of Perfect Checkpoints*

[image]
*First 100 points of Perfect Checkpoints*

[image]
*Last 100 points of Perfect Checkpoints*

3. Also, I tried using different minimize functions from `scipy`. The best configuration seems to be, using the minimizer `fminbound` and whenever the error comes out to be a large value, use `dual_annealing`. `dual_annealing` is more time consuming, therefore used only when the error comes out to be really large. The below image shows why the error comes out to be really large in some cases, the red line shows the corresponding point from perfect and generated, black line shows the one that has a wrong correspondence.

[image]
*Problem with using `fminbound` alone*

4. The following are the results using PilotNet and Explicit brain.

[image]
*Explicit Brain*

[image]
*PilotNet Brain*

SakshayMahna, 2021-11-28:
Hello!

I have finally resolved all the bugs and made some optimizations. I think this should work fine now. Here are some tests:

[image]
*Explicit Brain on Simple Circuit*

[image]
*PilotNet Brain on Simple Circuit*

[image]
*Explicit Brain on Many Curves Circuit*

[image]
*PilotNet Brain on Many Curves Circuit*

sergiopaniego, 2021-11-29:
The last perfect point shouldn't be separated from the path, that's a bug. 
Is the only point that doesn't follow the path? If so, we should just remove that last point from that bag and upload it a a new perfect bag.
I will review all the perfect bags searching for these kind of points that are out of the path (probably just the last points should have this problem).

sergiopaniego, 2021-11-29:
The rest of the PR looks great!! Solving that small problem with the perfect bags we could accept the new metric calculation
What about the metric calculation time?

sergiopaniego, 2021-11-29:
I have updated the simple circuit perfect checkpoints bag removing that last point.
Could you run the experiments for the different brains again? You could add opencv brain too to have more results.

SakshayMahna, 2021-12-02:
These are the results:

[image]
*OpenCV Brain*

[image]
*PilotNet Brain*

[image]
*Explicit Brain*

The above observations do seem correct. The explicit performs the best, there are some small oscillations in PilotNet and in OpenCV brain, there were quite a lot smooth oscillations, and the car was not following the track much consistently. What do you think @sergiopaniego?

Also, I noted the time for these circuits:

- Simple Circuit ~ 30 seconds
- Many Curves ~ 1 minute
- Montmelo ~ 30 seconds

sergiopaniego, 2021-12-02:
Sounds great to me!
Reviewing the code, lap_point is no longer needed, so be can remove it from the function arguments. Additionally, since that argument is gone, we can call this function and obtain the metric without completing the lap, so we should add a few modification.
I will update the code with these small changes and then we should be good to go accepting the PR :smile:

sergiopaniego, 2021-12-02:
Ups! I have tried to generate the position deviation error for an uncompleted lap but it doesn't seem to work.
Do you think we could extend the code to cover those cases? That would be also interesting!

SakshayMahna, 2021-12-02:
Yes, currently, this does not work for partial laps. Need to look into how we can extend this.

SakshayMahna, 2021-12-03:
Hello!
I have extended the code for the cases of Over and Under completed laps. I was able to test the metrics for over completed laps directly. For the partial completed laps, I generated the ROSbag and tested on it, using a separate Python script. I think it should be working.

@sergiopaniego could you check it once for the under completed laps?

sergiopaniego, 2021-12-03:
I have tested it and work fine!!
You can now apply the changes from https://github.com/JdeRobot/BehaviorMetrics/pull/261 to get rid of lap_point on the function arguments and add the visualization of the metrics when the lap is not completed, as we discussed yesterday. I think the changes are small. Once you complete them, we can finally merge it :smile: !!!

SakshayMahna, 2021-12-03:
Yes, I have applied the changes from the recent PR, and checked the working as well!

sergiopaniego, 2021-12-08:
Thanks for your great work @SakshayMahna!

closed 2021-12-08 by sergiopaniego