JdeRobot/BehaviorMetrics #268: [F1] Position Deviation failing on certain cases

SakshayMahna, 2021-12-17:
The position deviation metric is failing on some cases. To correct the errors and make the calculation more robust.

SakshayMahna, 2021-12-17:
An example case:

[image]

The position deviation metric keeps on running indefinitely for this case.

SakshayMahna, 2021-12-17:
The position deviation works by iterating over the points of the perfect bag and spline parameter `t`. This parameter `t` starts from 0 to some value `t_max` which we already know. It is the number of points that are collected while evaluating the robot car.

As we progress through the iteration loop, the value of `t` keeps on increasing from 0 to `t_max`. The loop is supposed to halt when `t` has a value greater than `t_max`. However, in this case, it seems that `t` does not cross the `t_max` value. As can be seen in the figure below. The print statements can be read as `Bad <calculated_distance> <t> <t_max>`. As can be seen, in the highlighted text, `t` never exceeds `t_max`, and hence the loop is stuck.

[image]

*Print Statements*

Another figure describing the same issue. The points of the perfect bag keep on iterating, however, the spline points have stopped their iteration. Yellow points are the points from perfect bag and blue points are the points from the spline.

[image]

*All further points of perfect bag are associating themselves with a single point on the spline*

A perfect use case would have been.

[image]

*Both the perfect bag points and spline points stopping at the same time*

SakshayMahna, 2021-12-17:
As a solution, I think we can add another termination condition, more like an error interrupt. That whenever the iterations are converging to a same point on the spline. Then, this is not an expected behavior and we should stop our execution.

I'll prepare a draft code for the same. So, we can test on other different bags as well that are giving the same kind of problems.

SakshayMahna, 2021-12-17:
@sergiopaniego Let's try with this termination condition.

sergiopaniego, 2021-12-20:
Great solution!
I have tested it and it returns results that seem legit, so I think we can merge it. 
I will again test it with lots of experiments and give you further details
Again, thank you for this great solution and for your quick response :smile:

SakshayMahna, 2021-12-20:
Thank you Sergio!

I will also keep on testing from my side and update regarding any bugs.

SakshayMahna, 2021-12-21:
For some tests I could observe that the difference calculation was not taking place from start to end. Some parts towards the end were missing. 

In order to correct that, I changed the spline boundary conditions from natural to clamped. According to documentation, natural spline implies the end points have their second derivative as zero and clamped implies that the end points have their first derivative zero. On doing some tests, the clamped version seems to give the same performance along with compensating for the above mentioned issue. Also, it would make sense to use clamped as this is the scenario of the simulation as well, the car starts with zero velocity and ends with zero velocity.

closed 2021-12-27 by sergiopaniego