JdeRobot/BehaviorMetrics #333: [Doc] Updating Installation of ROS Noetic and QuickStart

laukikk, 2022-04-18:
## Installation of ROS Noetic:
- Incorrect order for installation of BehaviourMetrics and gym-gazebo
- For manual installation, some more clarity on the folder placements of BehaviourMetrics, CustomRobots, and Sophus
- behavior_studio is now behavior_metrics

## QuickStart
- Running applications require ~python~ `python3`
- Brains Tutorial is missing (Already present at #48)

sergiopaniego, 2022-04-19:
Thanks for pointing out some possible improvements.
Most of them are small improvements (except the last one). Could you open the PR with this modifications?

laukikk, 2022-04-20:
Yes, I'm working on that. Also regarding #48, I'll also make a PR for it soon.

laukikk, 2022-05-08:
@sergiopaniego, this PR has been merged so it can be safe to close dthis issue now.

PR #334 (2022-05-06): Doc changes of ROS Noetic installation and Quick Start guide
The following changes would be reflected:

## ROS Noetic Installation Docs:
- [x] Spelling check
- [x] changes in the manual installation of ROS for [ROS Wiki](http://wiki.ros.org/noetic/Installation/Ubuntu)
- [x] VNC for chrome is deprecated mentioned
- [x] behavior_studio is now behavior_metrics

## Quick Start guide
- [x] Commented the link for the tutorial for "Train a Brain". (Yet to develop)
- [x] bash commands update

this commit fixes #333