JdeRobot/BehaviorMetrics #353: [Bug] GUI crashes when the gazebo button on the Simulation subsection is clicked

laukikk, 2022-05-19:
The error:
``` bash
Traceback (most recent call last):
  File "/root/BehaviorMetrics/behavior_metrics/ui/gui/views/toolbar.py", line 259, in mousePressEvent
    self.parent.open_close_simulator_gui()
  File "/root/BehaviorMetrics/behavior_metrics/ui/gui/views/toolbar.py", line 721, in open_close_simulator_gui
    if not environment.is_gzclient_open():
  File "/root/BehaviorMetrics/behavior_metrics/utils/environment.py", line 105, in is_gzclient_open
    ps_output = subprocess.check_output(["ps", "-Af"]).strip("\n")
TypeError: a bytes-like object is required, not 'str'
Aborted
```

sergiopaniego, 2022-05-19:
Thanks for the report!
Do you have a suggestion to fix it?

laukikk, 2022-05-19:
Due to the updates in the packages, I guess the default parameters of the functions were changed. I'll be making the PR in some time.

sergiopaniego, 2022-05-20:
Thanks for the contribution!

PR #354 (2022-05-20): Fixed gazebo crashing issue
this commit fixes #353.
closed 2022-05-20 by sergiopaniego