JdeRobot/BehaviorMetrics #396: Issue using Pytorch quantized models

nik1806, 2022-08-24:
Traceback (most recent call last):
  File "driver.py", line 221, in <module>
    main()
  File "driver.py", line 197, in main
    pilot = Pilot(app_configuration, controller, app_configuration.brain_path)
  File "/root/BehaviorMetrics/behavior_metrics/pilot.py", line 69, in __init__
    self.initialize_robot()
  File "/root/BehaviorMetrics/behavior_metrics/pilot.py", line 115, in initialize_robot
    self.configuration.experiment_model, self.configuration.brain_kwargs)
  File "/root/BehaviorMetrics/behavior_metrics/brains/brains_handler.py", line 28, in __init__
    self.load_brain(brain_path)
  File "/root/BehaviorMetrics/behavior_metrics/brains/brains_handler.py", line 55, in load_brain
    self.active_brain = Brain(self.sensors, self.actuators, model=self.model, handler=self, config=self.config)
  File "/root/BehaviorMetrics/behavior_metrics/brains/f1/brain_f1_torch.py", line 64, in __init__
    self.net = torch.jit.load(PRETRAINED_MODELS + model).to(self.device)
  File "/usr/local/lib/python3.7/dist-packages/torch/jit/_serialization.py", line 161, in load
    cpp_module = torch._C.import_ir_module(cu, str(f), map_location, _extra_files)
RuntimeError: 
Unknown type name 'NoneType':
Serialized   File "code/__torch__/torch/nn/quantized/modules/linear.py", line 14
    return (qweight, bias, training, dtype)
  def __setstate__(self: __torch__.torch.nn.quantized.modules.linear.LinearPackedParams,
    state: Tuple[Tensor, Optional[Tensor], bool, int]) -> NoneType:
                                                          ~~~~~~~~ <--- HERE
    self.dtype = (state)[3]
    _0 = (self).set_weight_bias((state)[0], (state)[1], )

nik1806, 2022-08-29:
The problem was with installed Pytorch version. I upgraded it to 1.11.0 and everything worked as expected.

sergiopaniego, 2022-08-31:
We should update the requirements.txt then, right?

nik1806, 2022-08-31:
Yes, we can update to latest work of PyTorch.

sergiopaniego, 2022-08-31:
Great! Can you open the new PR, please?

PR #400 (2022-08-31): Update PyTorch requirements
fixes #396