JdeRobot/BehaviorMetrics #411: Use CarlaBirdEyeView from repo instead of pip

sergiopaniego, 2022-12-13:
Pip installation requires numpy==1.18.0 and this version has some bugs. 
We could download the repo and install them from scratch to avoid this problem.

I have asked the developer of the repo about this issue on the original repo: https://github.com/deepsense-ai/carla-birdeye-view/pull/12

GOVINDFROMINDIA, 2023-02-16:
To download the repository and install it from scratch, you can follow these general steps:

Clone the repository to your local machine: git clone https://github.com/deepsense-ai/carla-birdeye-view.git
Navigate to the repository directory: cd carla-birdeye-view
Create a virtual environment to isolate the installation: python -m venv venv
Activate the virtual environment: source venv/bin/activate (on Linux/macOS) or venv\Scripts\activate (on Windows)
Install the required dependencies: pip install -r requirements.txt
Run the program as described in the repository's documentation.

Please assign me this issue

sergiopaniego, 2023-02-17:
We have solved this issue using the flag: 

`--no-deps`

when installing via pip. You could add it to the current documentation on the webpage.