JdeRobot/BehaviorMetrics #494: Build error while trying to install CARLA ROS Bridge from source

nik1806, 2023-01-28:
During step 7 from the instructions - https://carla.readthedocs.io/projects/ros-bridge/en/latest/ros_installation_ros1/#b-using-the-source-repository

I got the following error:
```
####
#### Running command: "make build -j32 -l32" in "/home/nikhil/carla-ros-bridge/catkin_ws/build"
####
make: *** No rule to make target 'build'.  Stop.
Invoking "make build -j32 -l32" failed
```

The required files are created but as expected when trying to run ros bridge it fails.

What could be the solution for the issue?

sergiopaniego, 2023-01-30:
Hello! 

Could you give further details on the exact command of the installation that leads to the error posted?
Have you fulfilled the prerequisites for installation as described in the document [https://carla.readthedocs.io/projects/ros-bridge/en/latest/ros_installation_ros1/#before-you-begin](https://carla.readthedocs.io/projects/ros-bridge/en/latest/ros_installation_ros1/#before-you-begin)?

nik1806, 2023-01-31:
Hello!
The following command (step 7 from `B. Using the source repository`) led to the error:
```
catkin_make build
```

I have checked the prerequisites. I have
ROS Neotic,
rviz, and
CARLA_0.9.13

nik1806, 2023-01-31:
However, to install the requirements of Behavior Metrics, I upgrade my python version to 3.8 (to solve dependency conflict).
I found that CARLA version is for Python 3.7. Could this be related to the issue?

sergiopaniego, 2023-01-31:
I think that the problem should not come from the python upgrade. I have Behavior Metrics in several machines installed, all of them with python 3.8 and they have support for CARLA too. 

Have you tried with `catkin build `instead of `catkin_make build`?

nik1806, 2023-02-02:
Hello!

Somehow I didn't have catkin installed. The build was successful after I installed catkin with pip. 
Thank you!

closed 2023-02-02 by nik1806