JdeRobot/BehaviorMetrics #581: Launching Drone using drive_gazebo error no module names 'drone_wrapper_class'

damodardatta, 2023-03-11:
I followed the instructions for installation[ here](https://jderobot.github.io/BehaviorMetrics/install/). 

additionally installed: sudo apt-get install ros-noetic-jderobot-drones

but getting error :

File "/home/damodar/GSOC/JDE/BehaviorMetrics/behavior_metrics/brains/drone/brain_drone_explicit.py", line 4, in <module>
    from drone_wrapper import DroneWrapper
  File "/opt/ros/melodic/lib/python2.7/dist-packages/drone_wrapper/__init__.py", line 1, in <module>
    from drone_wrapper_class import DroneWrapper
ModuleNotFoundError: No module named 'drone_wrapper_class'

Dhruv2012, 2023-03-13:
Hi,
I am able to run the example script for car using```python3 driver_gazebo.py -c ./configs/default.yml -g```. 

However, for the drone, I am also facing the same error as above.

sergiopaniego, 2023-03-13:
Since the version we are targeting is ROS Noetic, the command should be: 
`sudo apt-get install ros-noetic-jderobot-drones`

The current version of the installation webpage is outdated. You could open a PR with the installation instructions updated once you manage to install it so you start contributing to the project :smile:

damodardatta, 2023-03-14:
@sergiopaniego I found that "simple_circuit_drone.launch" does exist in '/opt/jderobot/share/jderobot/gazebo/launch/'. I have checked the same in docker image also. Can you check it on your side?

sergiopaniego, 2023-03-15:
Thanks for pointing that out.
You can check the drones repository and install it accordingly: https://github.com/JdeRobot/drones. It's not currently shown in the installation instructions so we need to add it (you can open a PR with the fix)
We removed the drones packages from the current docker image to reduce the size i. If you go back a to the previous commit (https://github.com/JdeRobot/BehaviorMetrics/blob/6a4e1cd7b3799e32c463bcc02e7bc085b0748f9d/.docker/noetic/Dockerfile)   you can find the commands used to install the drone packages if needed

damodardatta, 2023-03-21:
Thanks, Sergio was able to launch the drone after installing it in noetic with (branch: noetic-devel instead of melodic)
```
sudo apt-get install ros-noetic-jderobot-drones 
```
 Yeah, I will do a PR to document it properly for installation. So others can use it.

closed 2023-03-21 by damodardatta

yashphalle, 2023-03-31:
I was executing this command  for launching drone 
`python3 driver_gazebo.py -c configs/default-drone.yml -g`
It is showing world with f1 track but drone is not spawning.
Is this command correct or what else I am missing ?

damodardatta, 2023-04-01:
Have you checked if inside px4 you are able to launch the drone? If yes then check if the symlinks are working, this is a bit cumbersome as you have to check each individual component.