JdeRobot/BehaviorMetrics #584: Installation: package dependency conflicts

tejalbarnwal, 2023-03-13:
As stated in the [requirements. txt](https://github.com/JdeRobot/BehaviorMetrics/blob/noetic-devel/requirements.txt), `carla-birdeye-view 1.1.1` depends on `numpy==1.18.3`, but other modules like `tensorflow-gpu 2.11.0` depend on `numpy>=1.20`, which causes conflicts while running `requiurements.txt` file. Is there any way to upgrade the numpy dependency for carla-birdeye-view?

The issue has been previously discussed [here](https://github.com/JdeRobot/BehaviorMetrics/issues/43#issuecomment-1465823890), where @sergiopaniego recommend the following:
>Since carla-birdeye-view seems to be no longer in development, some packages could be old versions. I have tested with newer versions of numpy and they don't crash at any point, so it seems legit installing the latest numpy version. To overcome this issue during the installation, you can add --no-deps to the pip install -r requirements.txt command.

sergiopaniego, 2023-03-14:
Thanks for opening the issue. I think that a possible workaround could be adding `--no-deps` to the packages installation instruction. What do you think? At least for that specific package.
[https://stackoverflow.com/questions/12759761/pip-force-install-ignoring-dependencies](url)
This solution should be tested in a new virtualenv before we can proceed with it.

tejalbarnwal, 2023-03-14:
* To install the `carla_birdeye_view==1.1.1` with `--no-deps` flag, I guess we would have to run `pip3 install --no-deps carla_birdeye_view==1.1.1` separately from `pip3 install -r requirements.txt `.
I tested the above steps where I first ran `pip3 install -r requirements.txt` followed by `pip3 install --no-deps carla_birdeye_view==1.1.1`.  Also, for this, I removed the `carla_birdeye_view==1.1.1` requirement from `requirements.txt`

* I also tried the option `pip install --no-deps -r requirements.txt` in a virtual environment, but that resulted in ignoring the dependencies of other packages as well.
I tried finding if it was possible to specify `--no-deps` for a specific requirement in `requirements.txt`; I found [here](https://github.com/pypa/pip/issues/7339), that this isn't currently supported. 

Shall I open a PR for the former method I specified above?

sergiopaniego, 2023-03-15:
Sure! Sounds good to me. We need to update at least :
* Installation instructions in webpage (probably a separated PR pointing to gh-pages branch)
* Dockerfiles.
* requirements.txt

PR #589 (2023-03-20): resolve: pip package dependency conflicts
Closes #584 
## Summary
* Removed the line `carla_birdeye_view==1.1.1` from `requirements.txt` 
* Updated Dockerfiles with following statements:
  ```
  pip3 install -r requirements.txt
  pip3 install --no-deps carla_birdeye_view==1.1.1
  ```
closed 2023-03-20 by sergiopaniego