JdeRobot/BehaviorMetrics #587: time-out of 10000ms while waiting for the simulator

Rhythmbellic, 2023-03-15:
I downloaded the behaviour metrics and tried to run it
but after launching, simulator is terminating in seconds and showing error
(20: SimulatorEnv: launching simulator server.
20: SimulatorEnv: launching simulator server.
Traceback (most recent call last):
  File "driver_carla.py", line 353, in <module>
    main()
  File "driver_carla.py", line 283, in main
    controller = ControllerCarla()
  File "/home/rhythm/catkin_ws/CustomRobots/f1/build/Sophus/build/BehaviorMetrics/behavior_metrics/utils/controller_carla.py", line 77, in __init__
    self.world = client.get_world()
RuntimeError: time-out of 10000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000
).

Rhythmbellic, 2023-03-15:
ok I have solved this problem-
1)go to the path where you have installed carla_simulator.
2)open the path in terminal.
3)Run the command in terminal
./CarlaUE4.sh
4) you will get some extra dependencies you haven't installed
5) Install it and your carla will run

sergiopaniego, 2023-03-15:
This problem should be solved following the installation instructions on the webpage. 

Closing the issue

closed 2023-03-15 by sergiopaniego