JdeRobot/BehaviorMetrics #590: BirdView_V2_cache files are missing

Rhythmbellic, 2023-03-17:
Can anyone help that it's because of wrong behavior metric installation or bird-eye-view installation
that  I am getting error No such file or directory: 'birdview_v2_cache/Carla/Maps/Town01__px_per_meter=10__opendrive_hash=b5c4219c4d5e337c5ba2c2d7dc9cb2ef3b417bec__margin=300.npy.lock'
the map files are not in the folder and I have cloned Behavior Metrics file again  to verify the installation. 
Pls help me to sort this error

sergiopaniego, 2023-03-17:
Hi! 
This error was raised because that folder should be created before running the application. I'm creating a PR to fix it

sergiopaniego, 2023-03-17:
You can now pull the latest changes and test it again

PR #591 (2023-03-17): Update location of .gitignore file
fix #590
closed 2023-03-17 by sergiopaniego