JdeRobot/BehaviorMetrics #598: driver_carla.py RuntimeError: time-out of 10000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000

Dhruv2012, 2023-03-28:
Hi,
I am facing this issue while executing `driver_carla.py`. 

Below is the terminal output:

```
40: CARLA is not supported
Failed to load Python extension for LZ4 support. LZ4 compression will not be available.
Python older than 3.7 detected. 
20: SimulatorEnv: launching simulator server.
20: SimulatorEnv: launching simulator server.
Traceback (most recent call last):
  File "driver_carla.py", line 353, in <module>
    main()
  File "driver_carla.py", line 283, in main
    controller = ControllerCarla()
  File "/home/dhruv/jderobot_gsoc23/BehaviorMetrics/behavior_metrics/utils/controller_carla.py", line 77, in __init__
    self.world = client.get_world()
RuntimeError: time-out of 10000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000
```

```
**Hardware:**
4GB Nvidia 940MX + i7 8th Gen

**OS**
Ubuntu Linux 18.04
CARLA 0.9.13
```
As this happens to be related to CARLA, I directly ran the stand-alone CARLA server using `./CARLAEU4.sh`. However, it crashes into a black screen. 

Is there a **docker image** of **Behavior Metrics** having **CARLA** as well, that I can possibly use? Or any suggestions on optimizing configuration of CARLA (graphics or more)?

Thanks!

sergiopaniego, 2023-03-28:
You can check the logs output for detailed information. 
The server-client are launched from environments.py file as subprocesses and they generate log files. (https://github.com/JdeRobot/BehaviorMetrics/blob/noetic-devel/behavior_metrics/utils/environment.py#L58 and https://github.com/JdeRobot/BehaviorMetrics/blob/75e88509cb7a6128f7d7aa0f3f12a1b1e05b2ffe/behavior_metrics/utils/environment.py#LL67C24-L67C24) 

Your machine is not meeting the requirements described by CARLA. 
You could also check if you can run Carla with low-level graphics: https://carla.readthedocs.io/en/latest/adv_rendering_options/#quality-levels. 

We don't have a docker image with CARLA support, but you can use the ones we already provide and install CARLA in the container using the installation commands as usual. 

Could you obtain a more powerful machine? Cloud service?

Rhythmbellic, 2023-03-29:
Hey Dhruv observe first 3 lines of error carefully

''40: CARLA is not supported
Failed to load Python extension for LZ4 support. LZ4 compression will not be available.
Python older than 3.7 detected.''
your python version is older than required version try removing it and installing latest version,
It worked for me after that.

Dhruv2012, 2023-03-30:
Hi @Rhythmbellic
Sure. Will try that out.

sergiopaniego, 2023-03-31:
Is this issue solved?

sergiopaniego, 2023-03-31:
@Dhruv2012

Dhruv2012, 2023-04-01:
Hi,
Unfortunately, it did not work. Before launching the client, the server itself crashes, before coming to the step where python version is utilized to call client script. I have temporarily arranged an alternate machine where CARLA is working.

sergiopaniego, 2023-04-03:
Good solution! We can mark this issue as solved.

closed 2023-04-03 by sergiopaniego