JdeRobot/BehaviorMetrics #601: Creating New Drone Scenerio

yashphalle, 2023-03-29:
Need to create new scenarios like Drone Gymkhana to work on 3-D Control of drone.

Started working on creating world in Gazebo.

Opened as sub-part of Issue [Add drone_gymkhana.world from RoboticsAcademy exercises drone_gymkhana for testing drone 3d control #594]

jmplaza, 2023-04-09:
Hi @yashphalle , for scenarios just create them at the [CustomRobots repository](https://github.com/JdeRobot/CustomRobots) , the ROS-Noetic branch. It is going to be refactored soon as RoboticsInfrastructure repository but it will have a CustomRobots directory for the robot models, worlds, plugins to be used from different projects (such as BehaviorMetrics, RoboticsAcademy, VisualCircuit...)

yashphalle, 2023-04-10:
Okk Sure Sir!

closed 2023-04-10 by yashphalle