JdeRobot/BehaviorMetrics #602: TODO for Project #8: Obstacle avoidance for autonomous driving in CARLA using segmentation deep learning models.

Dhruv2012, 2023-03-30:
Roadmap for **GSoC 23 - Obstacle avoidance for Autonomous Driving in CARLA using segmentation deep learning models**

- [ ]  Setting up and getting familiar with relevant Software (Behavior Metrics, CARLA, etc.) 
  - [x]  Behavior Metrics
  - [ ]   DeepLearning Studio (**IN PROGRESS**)
  - [ ]  CARLA (**IN PROGRESS**) 
- [ ]  Developing **obstacle avoidance behavior using range-based methods**  (**IN PROGRESS**)
        - Get familiar by implementing basic obstacle avoidance behavior using Virtual Force Field algorithms
- [ ]   Obstacle avoidance using ground truth semantic mask from CARLA
         - Utilize ground truth semantic mask from CARLA to set goal points (per-frame) and feed it to a force field-based obstacle avoidance algorithm
- [ ]   Semantic segmentation for Autonomous Driving
         - Train and evaluate semantic segmentation networks on CARLA simulator data.
- [ ]   Semantic segmentation + Obstacle avoidance
         - Repeat this similar to ground truth, but instead use predicted semantic mask to set local goal points.
 
- [ ]   End-to-end learning
         - Propose an end-to-end approach, where a network can learn to predict semantic mask and control inputs, when fed with robot pose and input image.

jmplaza, 2023-03-30:
Hi @Dhruv2012 , 

this issue seems the roadmap for your application to GSoC-2023, not something to be done at the BehaviorMetrics tool itself... There are other people also working on the tool. Creating an issue in the repository of the tool for your application is not a good idea, it is not the best place for that :-)

sergiopaniego, 2023-03-30:
Hi @Dhruv2012,

This roadmap should be part of the application document only (although it looks good) . In here, we should open small issues (tasks) that we are fixing using pull requests. For example, an issue could be "Launching CARLA maps with an ego vehicle and one obstacle" where you generate the code to launch the CARLA world with an obstacle, based on the current codebase. 

Would you like to open your first issue and generate the pull request based on these ideas? :smile:

closed 2023-03-30 by sergiopaniego

Dhruv2012, 2023-03-31:
> Hi @Dhruv2012 ,
>
> this issue seems the roadmap for your application to GSoC-2023, not something to be done at the BehaviorMetrics tool itself... There are other people also working on the tool. [...]

Sure, @jmplaza. Apologies. Would take care!