JdeRobot/BehaviorMetrics #603: Add a launch for Town 01 with parked vehicles

Qi-Zha0, 2023-03-30:
I would like to create a launch file with a stopped vehicle to prepare for obstacle avoidance learning tasks. 
Expected outcome: a launch file that loads the world Town 01 with an ego vehicle and one or more parked vehicles in CARLA

Qi-Zha0, 2023-03-31:
@sergiopaniego Thank you for the tip! #606 I added the launch file as well as an object definition file for spawning all required objects. I noticed currently we are loading the [default Tesla Model 3](https://github.com/carla-simulator/ros-bridge/blob/master/carla_spawn_objects/config/objects.json) from `carla_spawn_objects`, I created a custom .json file with an extra vehicle to be spawned. However, since it is out of the `carla_spawn_object` package, the launch file can no longer find it by `find package`, the best work-around I could come up with is to use an environmental variable.

To use the launch file, 
`export OBJECT_PATH="/path/to/behavior_metrics/configs/CARLA_launch_files/CARLA_object_files"`
`python3 driver_carla.py -c configs/default_carla_parked_vehicle.yml -g `

Note: In `controller_carla.py`, the ego vehicle is selected by choosing the first vehicle from all vehicles spawned, to make sure Tesla is always selected as the ego vehicle, use `filter('vehicle.tesla.*')`

Example: Tesla as the ego vehicle and a stopped Audi A2 without any sensors: 
[image]

sergiopaniego, 2023-04-03:
New issue created: https://github.com/JdeRobot/BehaviorMetrics/issues/608. If you will, you can generate a new PR with the updated documentation. We can discuss it via the new issue.

closed 2023-04-03 by sergiopaniego