JdeRobot/BehaviorMetrics #604: app crashing while loading the simulation

Rhythmbellic, 2023-03-30:
The app is crashing again and again when I am trying to load the inbuilt simulation and also it don't have option for loading a new simulation pls help.
https://user-images.githubusercontent.com/92723976/228941101-7ef019a6-3b95-49c9-8631-5d03f851d2ff.mp4

sergiopaniego, 2023-03-31:
Thanks for opening the issue. The issue is related to loading another world. Since the project started with support for Gazebo and then we added CARLA, some functionalities are still not ported to CARLA. The functionality for loading a new world is not yet available for CARLA. 
It would be a nice issue to cover with a pull request. Would you like to work on it?

Rhythmbellic, 2023-03-31:
Sure I will go through both carla driver. Py and gazebo carla. Py for reference is any other hint you want to give before start working on this issue?

Rhythmbellic, 2023-04-08:
hey @sergiopaniego  I have a query, I have learned pyqt5 and gone through the GUI files right now the brain is launching from the carla ros bridge launch file and looking for the solution but I got a new issue, currently our app launch world from ros bridge file, shouldn't we load the world launch files from config folder?

Rhythmbellic, 2023-04-08:
just tell me so i can set the path as our problem have 3 changes according to me

- [ ] Create a Carla launch file to launch the Carla simulator with the correct settings 

- [ ] Modify the world_selection.py file in the BehaviorMetrics module to include the load_carla_world function and modify the load_world function to use it if the simulator selected by the user is "Carla".

- [ ]

sergiopaniego, 2023-04-12:
Hi @Rhythmbellic!

We use CARLA ROS Bridge because we provide support to Gazebo too and the application structure follows the same procedures for both simulators (Gazebo uses ROS too), so we need to launch the world using the current pipeline. 
The load world functionality should follow a similar approach.

I hope this answers your query.