JdeRobot/BehaviorMetrics #613: Town01 scenario with dynamic objects

Qi-Zha0, 2023-04-04:
#603 creates a scenario with one ego vehicle and one stopped vehicle. I am working on a more complex scenario in Town01 with multiple dynamic objects for the obstacle avoidance task.

The expected outcomes are:

- new launch file for the scenario with one ego vehicle and multiple AD agents such as pedestrians, cars, and bikes
- related object JSON file
- updated documentation on how to launch the example

Qi-Zha0, 2023-04-07:
Investigating why starting the simulation from the GUI after generating traffic causes the system to crash with error
```
30: ValueError
30: ERROR Pilot Carla!
```
[roslaunch_stdout.log.txt](https://github.com/JdeRobot/BehaviorMetrics/files/11175240/roslaunch_stdout.log.txt)

when `pilot_carla.py` calls the brain's `execute()` function.

The log file looks normal.

sergiopaniego, 2023-04-12:
Hi @Qi-Zha0! 
Thanks for opening the issue. The current system is designed for one agent, so this addition could be a little bit tricky. 

Could you give further details about the issue with the `bird_eye_view` functionality? When talking about reading images, are you talking about [getImage](https://github.com/JdeRobot/BehaviorMetrics/blob/82417f49f848c6dbaf52545a9a6c696a52ffbdaa/behavior_metrics/robot/interfaces/birdeyeview.py#L16) functionality?

Qi-Zha0, 2023-04-14:
@sergiopaniego 
Yes, it's the [birdview_producer.produce()](https://github.com/JdeRobot/BehaviorMetrics/blob/82417f49f848c6dbaf52545a9a6c696a52ffbdaa/behavior_metrics/robot/interfaces/birdeyeview.py#L17) call th
at causes the problem. I tried two ways of generating other vehicles and pedestrians:

1) adding more objects to the object JSON file and spawning them in the launch file
2) use CARLA Python API in a script to spawn vehicles and walkers and run it separately 

In both cases, the moment the vehicles and pedestrians are spawned, the simulation terminates with the aforementioned error. 

Here is an image of a biker autonomously driving ahead of the ego vehicle in manual control mode. As soon as the bird eye view deep learning model is loaded, the error occurs. 

[image]

sergiopaniego, 2023-04-17:
Thanks for the update, @Qi-Zha0.

Would you happen to have a branch with the current changes so we can discuss it based on the code directly? You could create a work-in-progress PR with your changes so we can discuss it using code. 

We have added a few config files with other vehicles: https://github.com/JdeRobot/BehaviorMetrics/pull/614. These config files are a good starting point for the generalization that we pursue.

Idea:
Include a `print(ex)`  where the exception is controlled to understand the issue further.
https://github.com/JdeRobot/BehaviorMetrics/blob/ce40281cd497d67449ebcb85531332916714e7c3/behavior_metrics/pilot_carla.py#L147

Qi-Zha0, 2023-04-18:
@sergiopaniego Thank you for the debugging tip! I figured out how to properly set up a vehicle with an ad agent in the launch file. I wasn't doing it correctly to add the ad agent for the npc. I have created a WIP pull request #615 with a new scenario with the ego vehicle and an npc car. 

To view the scenario alone, run `roslaunch town_01_anticlockwise_single_ad_npc.launch` after starting CARLA, where the npc car would be initialized next to the ego vehicle and then drive away. The problem is when starting the vehicle in the BM simulation, the npc car doesn't move. I'm investigating this issue currently and would appreciate any advice or idea

sergiopaniego, 2023-04-19:
Nice work @Qi-Zha0!

Are you getting any errors when starting the vehicle in BM? 

I see a few points that need to be covered in order to accept the PR:

* The npc vehicle should start moving when the play button is pressed. When I launch BM using `python3 driver_carla.py -c configs/default_carla_single_ad_npc.yml -g`, the simulator and app launch and the npc vehicle immediately starts moving in the other lane (Audi car). The ego vehicle (Tesla) remains still. 
[image]

* There is some problem with the application guessing which is the main vehicle in the scenario. When the play button is pressed, sometimes the current main vehicle is actually the npc instead of the ego vehicle. The one that receives the commands and produces the sensor data is the npc. It sometimes works correctly but seems to be a structural problem.

Are the points that I describe similar to what you described?

Qi-Zha0, 2023-04-25:
@sergiopaniego Thank you for pointing that out! I was looking into how to resolve the first issue too.

For the first issue, I think we might need to run the simulation in synchronous mode such that the simulation only steps when `tick()` is called. Then only call `tick()` when the start button is pressed. I will look into this later today to see what structural changes might be needed.

For the second issue, I believe it's because of [this line ](https://github.com/JdeRobot/BehaviorMetrics/blob/b4660cb3649fe06c399d5ca8cb7926e50249b91c/behavior_metrics/utils/controller_carla.py#L83) in `controller_carla.py`.

Qi-Zha0, 2023-04-26:
@sergiopaniego 
I figured out a workaround for the first issue: launch `carla_waypoint_publisher` for the NPC vehicle, separately from the rest of the launch file, when the simulation is first started. For the second issue, when selecting the ego vehicle in `controller_carla` and the brain file such that only the vehicle with `role_name` called `ego_vehicle` can be selected as the ego vehicle. 

I made a new pull request with the changes: #620 

To test it, run `python driver_carla.py -c configs/default_carla_single_ad_npc.yml -g`. You should see an npc vehicle 20 meters ahead of the ego vehicle upon launching BM, then after pressing start simulation, both vehicles should start moving. The ego vehicle follows waypoints published by CARLA while the ego vehicle is driven by the bird eye view deep learning model. 

Let me know what you think :)

sergiopaniego, 2023-05-03:
Hi @Qi-Zha0! 

Thanks for opening the PR. I'm trying to reproduce the code. I can't find the .json described here:

`<arg name="objects_definition_file" default='$(env OBJECT_PATH)/single_ad_car.json'/>`

Is this file missing?

Qi-Zha0, 2023-05-03:
@sergiopaniego Thank you for pointing it out. My apologies. I forgot to remove .json files from my .gitignore. I just added a new commit with the object file

sergiopaniego, 2023-05-05:
PR merged! #620

closed 2023-05-21 by Qi-Zha0