JdeRobot/BehaviorMetrics #619: Adding sample CARLA simulator data for Semantic segmentation-based avoidance using BehaviorMetrics

Dhruv2012, 2023-04-20:
The simulator data should contain:

- Ego vehicle's state (pose, orientation, image frame etc.)
- Ground truth semantic information
- World state (dynamic/state objects' pose, orientation, etc.)

divyanshAgarwal123, 2026-09-26:
Hi maintainers, I’d like to work on this AI/ML issue. I’ll first scope a minimal reproducible CARLA sample with synchronized RGB and semantic-segmentation frames, ego pose, and world-state metadata, document the recording/schema steps, and validate that BehaviorMetrics can consume it for avoidance evaluation. Could you please assign it to me? Thanks!

divyanshAgarwal123, 2026-09-29:
Hi maintainers, following up on the CARLA semantic-segmentation sample. Is this contribution still wanted, and does the minimal scope I proposed above look right? If so, could you assign the issue to me? I can adjust the sample format based on your guidance. Thanks!