JdeRobot/BehaviorMetrics #657: Unable to install Carla, Server Down.

KlrShaK, 2024-01-16:
### Issue Description

I am encountering difficulties while attempting to install the Carla Simulator using the provided procedures on their website. The problem seems to persist, as indicated by recent discussions in the GitHub repository's issues section ([CARLA GitHub Issue #6332](https://github.com/carla-simulator/carla/issues/6332)).

### Error Message

```bash
Err:8 http://dist.carla.org/carla focal InRelease                       
  Could not connect to dist.carla.org:80 (34.227.255.250). - connect (111: Connection refused)
```

### Request for Assistance

I am seeking alternatives or guidance to overcome this installation issue. Any suggestions or workaround strategies would be greatly appreciated.

sergiopaniego, 2024-01-18:
Hi!

Could you give more details about the step that causes this problem. Based on the issue and the description that you provide, it seems like the sever used for downloading the simulator could be down. It could be automatically solved so I would try launching the command at different times of the day. Let me know if the issue persists.

KlrShaK, 2024-01-20:
Hey,

This occurred while I was following the official install documentation on [BehaviourMetric website.](https://jderobot.github.io/BehaviorMetrics/install/). I used the link (https://carla.readthedocs.io/en/0.9.13/start_quickstart/) on the website to try and install CARLA. I encountered the  error while following these steps:

```
    sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 1AF1527DE64CB8D9
    sudo add-apt-repository "deb [arch=amd64] http://dist.carla.org/carla $(lsb_release -sc) main"
   
    sudo apt-get update # -----------> I get the ERROR while executing this line
    sudo apt-get install carla-simulator
```

As shown by the open issue on the  carla-simulator/carla repo that I linked in my previous comment. This problem have been persistent since March 2023. I tried launching it at different times and on different days and still got the same error

DeepDuke, 2024-01-20:
same issue

DeepDuke, 2024-01-20:
I also cannot visit ` http://dist.carla.org/carla` for installation. Actually I could do that several months ago...

sergiopaniego, 2024-01-24:
Since this problem is still around https://github.com/carla-simulator/carla/issues/6332, please install CARLA directly using the **B. Package installation** option. This option involves directly downloading the package from CARLA's GitHub repo.

@KlrShaK
@DeepDuke

closed 2024-02-12 by KlrShaK