JdeRobot/BehaviorMetrics #662: 20: Waiting for vehicles, stuck in loop

KlrShaK, 2024-02-05:
Hey I installed Carla as you described in #657 , It runs perfectly fine when I do `./CarlaUE4.sh`. However, when I try to run the  following examples in [Quick Start CARLA](https://jderobot.github.io/BehaviorMetrics/carla/quick_start/). 

```
behavior_metrics$ python3 driver_carla.py -c configs/CARLA/default_carla.yml -g
40: CARLA is not supported
20: SimulatorEnv: launching simulator server.
20: SimulatorEnv: launching simulator server.
Traceback (most recent call last):
  File "driver_carla.py", line 429, in <module>
    main()
  File "driver_carla.py", line 308, in main
    controller = ControllerCarla()
  File "/home/<user>/BehaviorMetrics/behavior_metrics/utils/controller_carla.py", line 75, in __init__
    self.world = client.get_world()
RuntimeError: time-out of 100000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000
```
which is similar to #598 #587 , I tried running Carla on its own using './CarlaUE4.sh', I got the following output: 
```
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0
```
The simulator loaded fine and I can move around. I also changed `sudo sysctl dev.i915.perf_stream_paranoid=0` which removed the warning. and also the error in BehaviorMetric somehow changed now when I run. **(THIS IS MOST RECENT ERROR)**
```
/behavior_metrics$ python3 driver_carla.py -c configs/CARLA/default_carla.yml -g
40: CARLA is not supported
20: SimulatorEnv: launching simulator server.
20: SimulatorEnv: launching simulator server.
20: Waiting for vehicles!
20: Waiting for vehicles!
20: Waiting for vehicles!
.
.
.
```
The error `20: Waiting for vehicles!` loops continuously
This issue is similar to #588 , but it doesn't help solving the issue as Carla on its own runs fine and doesn't show any dependencies issue.

sergiopaniego, 2024-02-06:
Hi,
Thanks for opening the issue. Based on the information you're sharing, there could be a compatibility issue. It could be CARLA server, CARLA simulator frontend (Python API), or even carla-ros-bridge as it says `CARLA is not supported`. Could you please share more details about which version of CARLA and carla-ros-bridge you've installed?

The log files should be in your machine's root path `/tmp`, not inside Behavior Metrics. Could you please double-check if it is present there?

KlrShaK, 2024-02-08:
Solved this issue by adding parameter `-carla-server` when running the command ` ./CarlaUE4.sh -quality-level=Low -preferNvidia -caral-server`. I have tested with the launch scripts of **carla-ros-bridge** `( roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch )` . I will close the issue after testing it with BehaviourMetric scripts

sergiopaniego, 2024-02-10:
Sure! I’ll close the issue but you can reopen it if the issue persists

closed 2024-02-10 by sergiopaniego

sergiopaniego, 2024-03-28:
We are using Carla 0.9.15 in this case.

What version of GPU do you have? Are you able to run CARLA with some of the examples they provide without carla-ros-bridge?

ABD-01, 2024-03-29:
Yes, I knew about the Carla version. This one was a PC with 2 GPUs (total of 18 GB), unfortunately, it is a shared system so I won't be changing the Carla version. And yes I am able to run Carla smoothly without ros bridge.
GPU Information:
[image]

I will update on this once I try running behaviour_metrics on my personal laptop (GTX 4GB 1650Ti).

sergiopaniego, 2024-03-29:
Ok! That behavior is a little bit weird. I suggest that you review the log output generated by the ROS Nodes to understand further the problem. 
https://github.com/JdeRobot/BehaviorMetrics/blob/b7333abef48b2c506fc428836f3d1ac83590e5e1/behavior_metrics/utils/environment.py#L89

ABD-01, 2024-03-29:
Tried different ways, and finally did `rm -rf ~/carla-ros-bridge`. And reinstalled again from the source.
Followed the above steps and it is working now!

[image]

sergiopaniego, 2024-03-29:
Nice!