JdeRobot/BehaviorMetrics #667: Review quick start guide for CARLA simulator and update it

sergiopaniego, 2024-02-26:
For this issue, we would need to review in depth the Quick start guide for CARLA (https://jderobot.github.io/BehaviorMetrics/carla/quick_start/) and update it accordingly for new users. 

There are images in this doc file that are for Gazebo, so it would be interesting to update them with CARLA simulator examples

AryanNanda17, 2024-02-27:
@sergiopaniego, I would like to work on the issue.

MihirGore23, 2024-03-08:
I am facing significant lags in running Carla smoothly due to limitations of computational capacity at my end. @sergiopaniego Can u suggest some way I can run Carla using any specific cloudGPU, or any lite-version of Carla?

sergiopaniego, 2024-03-11:
CARLA allows running on a low-quality level (https://github.com/JdeRobot/BehaviorMetrics/blob/4c89221d3263a641eba4db329bc4229c80599371/behavior_metrics/utils/environment.py#L79).
Are you running the simulation using a low-quality level?
What is your current hardware setup? Specially the GPU

ArthurWANGZHE, 2024-03-20:
I have recently installed CARLA and I would love to contribute to this issue

sergiopaniego, 2024-03-20:
Hi @ArthurWANGZHE !
I have also assigned this issue to you. We can use this space to discuss the development of this upgrade

ArthurWANGZHE, 2024-03-20:
While I try to record a video clip for the document, I find that my 'load' and 'reload' buttons don't work. I wonder if this happens only to me, or if something has gone wrong.

sergiopaniego, 2024-03-20:
The load and reload buttons are not currently working in the GUI. You may also address it as a new issue+PR. If you want to reload the world, you have to close the application and relaunch.

Safwannn89, 2025-11-10:
@sergiopaniego , can I contribute to this issue