JdeRobot/BehaviorMetrics #672: Add Instruction for Setting up CARLA and Behavior Metrics on CloudGPU

AryanNanda17, 2024-03-08:
Since CARLA Simulation requires a minimum of 6GB VRAM which is not common and not many users have access to it. Moreover, 
Carla Simulation on cloud GPU not only requires the server but also display and there is no proper guide to setting up CARLA on cloud, so there is a need to add instructions for setting up Behavior metrics on Carla Simulation using CloundGPU. 
@sergiopaniego could you please assign this issue to me and suggest a good and cheap cloud GPU provider?

sergiopaniego, 2024-03-11:
Assigned!

Have you tried changing the quality level to low? https://github.com/JdeRobot/BehaviorMetrics/blob/noetic-devel/behavior_metrics/utils/environment.py#L79
And is the -RenderOffScreen flag active when you run CARLA? https://github.com/JdeRobot/BehaviorMetrics/blob/noetic-devel/behavior_metrics/utils/environment.py#L79

This installation instruction is interesting as an addition. I've no suggestions regarding cloud GPU providers since the decision of which to use involves many variables at the same time. If you suggest some ideas we can use this discussion thread to choose an option.

AryanNanda17, 2024-03-12:
Hello @sergiopaniego, 

> Have you tried changing the quality level to low? https://github.com/JdeRobot/BehaviorMetrics/blob/noetic-devel/behavior_metrics/utils/environment.py#L79
> And is the -RenderOffScreen flag active when [...]

I have attempted to run CARLA with reduced quality settings, but the terminal crashes.

> If you suggest some ideas we can use this discussion thread to choose an option.

Let me do a little research on the best and cheapest cloud option available and then I will update.

KlrShaK, 2024-03-15:
Heyya,
If this issue is still open, I'd suggest looking into issue [137](https://github.com/carla-simulator/carla/issues/137) on running CARLA in cloud, additionally I would suggest to look into 
- https://carla.readthedocs.io/en/latest/build_docker/
- https://github.com/jbnunn/CARLADesktop

These would help you to find resources on installing CARLA on cloud. I personally don't have access to cloud compute, so I couldn't test it.

sergiopaniego, 2024-03-18:
Thanks for sharing the resources @KlrShaK

sergiopaniego, 2024-03-18:
@AryanNanda17 any progress on this issue?