JdeRobot/BehaviorMetrics #675: Fixing the load and reload button

ArthurWANGZHE, 2024-03-23:
I fond that the load and reload buttons in the ui intereface are currently not working. I plan to fix this issue

sergiopaniego, 2024-03-25:
I've assigned this issue to you and updated the label. 
To give you more context, these buttons work when the simulator is Gazebo since that was the primary simulator at the beginning of this project. It is no longer the primary simulator and we do the majority of development using CARLA since it is the standard simulator for autonomous driving.
The functionality for each of these buttons is slightly different so if you fix any of them you may generate the PR for that specific one. 
Reload -> Restore all the components in the simulation. The optimal way would be to restore the positions of each element without reloading the simulator completely.
Load -> For this case, we need to reload the simulator client completely.

ArthurWANGZHE, 2024-04-07:
Hi! I have made some progress. Now I can reload swicth the brain between "brain_deep_learning" and "brain_slow_and_turn" using reload button. However, I have encountered some trouble while trying to reload other brains. The errors are as follows:

```
20: Reloading brain... /home/arthur/BehaviorMetrics/behavior_metrics/brains/CARLA/brain_carla_segmentation_based_imitation_learning.py
Waiting for vehicle with role_name 'ego_vehicle'
Waiting for vehicle with role_name 'ego_vehicle'

20: Reloading brain... /home/arthur/BehaviorMetrics/behavior_metrics/brains/CARLA/brain_carla_autopilot.py
Brain could not be loaded!
vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
```

sharmajatinjs, 2025-12-07:
Thanks @jmplaza! I appreciate the update.

I have just submitted PR #722 which fixes the logic for the Load and Reload buttons as requested. I identified that the Reload button wasn't connected to the reset_simulation function and fixed the load_brain logic.

I look forward to your feedback!