JdeRobot/BehaviorMetrics #691: CARLA ROS Bridge

ElaheDlv, 2024-05-24:
Hi,
I have a problem while trying to use the reference instruction for installing the ROS bridge. When I try (sudo add-apt-repository "deb [arch=amd64] http://dist.carla.org/carla $(lsb_release -sc) main") I get this error message (W: Failed to fetch http://dist.carla.org/carla/dists/focal/InRelease  Could not connect to dist.carla.org:80 (34.227.255.250), connection timed out
W: Failed to fetch http://dist.carla.org/carla-0.9.14/dists/all/InRelease  Unable to connect to dist.carla.org:http:
W: Some index files failed to download. They have been ignored, or old ones used instead.) Can you please help me to fix this?

zebin-huang, 2024-05-24:
Hi @ElaheDlv, I have the same issue. The server seems to be down. Refer to @https://github.com/carla-simulator/carla/issues/7017

The issue was resolved by following the instructions provided by [this comment](https://github.com/carla-simulator/carla/issues/7017#issuecomment-1908462106). 

### 1. Install Required System Dependency

Before downloading CARLA, install the necessary system dependency:

```
sudo apt-get -y install libomp5
```

### 2. Download the CARLA 0.9.15 Release

Download the CARLA_0.9.13.tar.gz file (approximately 16GB) from the official release:

```
wget https://carla-releases.s3.us-east-005.backblazeb2.com/Linux/CARLA_0.9.15.tar.gz

# for 0.9.14
wget https://carla-releases.s3.us-east-005.backblazeb2.com/Linux/CARLA_0.9.14.tar.gz
```

### 3. Unpack CARLA to the Desired Directory

Unpack the downloaded file to `/opt/carla-simulator/`:

```
tar -xzvf CARLA_0.9.15.tar.gz -C /opt/carla-simulator/

# for 0.9.14
tar -xzvf CARLA_0.9.14.tar.gz -C /opt/carla-simulator/
```

### 4. Install the CARLA Python Module

Finally, install the CARLA Python module and necessary dependencies:

```
python -m pip install carla==0.9.15

# for 0.9.14
python -m pip install carla==0.9.14

python -m pip install -r /opt/carla-simulator/PythonAPI/examples/requirements.txt
```

ElaheDlv, 2024-05-24:
Hi @zebin-huang. Does the carla 0.9.14 or 0.9.15 works with the carla-bridge? I get this error message. 
[image]
As the carla server is down I was trying to use the source repository. (https://github.com/carla-simulator/ros-bridge) But I come across the fact that it will not work with 0.9.15 that was written in the documentation (https://jderobot.github.io/BehaviorMetrics/install/). Could you work with 0.9.14/0.9.15?

sergiopaniego, 2024-05-27:
Updating this file https://github.com/carla-simulator/ros-bridge/blob/master/carla_ros_bridge/src/carla_ros_bridge/CARLA_VERSION with the new version 0.9.14/15, it works :smile:

closed 2024-06-28 by ElaheDlv