JdeRobot/BehaviorMetrics #710: /first_image saved as empty or black image from ROS 2 bags

canveo, 2025-05-11:
When processing ROS 2 bags with analyze_bag.py, the images from /first_image are saved as empty (zero size) or fully black. As a result, the output PNGs in first_images/ are invalid or missing.

divyanshAgarwal123, 2026-09-26:
Hi maintainers, I’d like to work on this. I’ll reproduce it with a ROS 2 bag, trace `/first_image` through deserialization and the `CvBridge` conversion in `analyze_bag.py`, then add a focused regression check that verifies the saved PNG has the expected dimensions and non-zero pixel data. Could you please assign it to me? Thanks!

divyanshAgarwal123, 2026-09-29:
Hi maintainers, following up on this issue. PR #712 added the RGB topic, but its description says image extraction was left for later. Is the empty or black /first_image problem still outstanding? If so, could you assign this issue to me to investigate the remaining extraction path? Thanks!